EcmaTC53/spec #7: Can someone explain Digital.InputPullUpDown?

dtex, 2020-12-18:
PullUp and PullDown are known to me, but I'm not sure what this constant means.

I thought I knew, or at least thought I should know, but now that I really want to know Google is failing me.

phoddie, 2020-12-18:
It is present to support MCUs which have a PullUpDown mode on GPIO. ESP32 was the motivation with [`GPIO_PULLUP_PULLDOWN`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html?highlight=gpio_pullup_pulldown#_CPPv420GPIO_PULLUP_PULLDOWN).

dtex, 2020-12-18:
That was the clue (search term) I needed. Found this:

> Setting GPIO_PULLUP_PULLDOWN turns on both pull-up and pull-down resistors, which pulls the pin voltage to approximately VCC/2.

> Not useful for digital GPIO functions, but you can use this to bias the input when sampling an analog signal.

closed 2020-12-18 by dtex