EcmaTC53/spec #10: The speed of communication on the I²C bus.

fivdi, 2020-12-21:
The [`I2C`](https://github.com/EcmaTC53/spec/blob/master/docs/tc53.md#ic) class has a constructor option called `hz` for specifying the speed of communication on the I²C bus.

Let's assume that there are two peripherals connected to the I²C bus, peripheral A and peripheral B. Let's also assume that the maximum I²C clock frequency permitted for peripheral A is 400 kHz and the maximum I²C clock frequency permitted for peripheral B is 1 MHz. The maximum values are defined in the datasheets for the peripherals.

Assume also that `hz` is set to 400_000 when creating the `I2C` instance for peripheral A and that `hz` is set to 1_000_000 when creating the `I2C` instance for peripheral B.

Later, when `I2C` methods are called to communicate with peripheral B, the I²C bus clock will operate at a frequency of 1 Mhz. However, peripheral A is also connected to the bus and it is now being clocked at a frequency out of specification so its behavior will be undefined.

The I²C bus clock frequency shouldn't exceed the maximum permitted clock frequency of the slowest peripheral on the bus. It would be better if the I²C bus clock frequency was a global setting for the bus rather than an individual setting for each peripheral connected to the bus.

phoddie, 2020-12-21:
@fivdi. I understand what you are describing. It is a valid concern. If the bus is operating at a speed unsupported by a device, it has the potential to confuse that device.

This sentence in the second paragraph of the introduction is relevant:

> Hardware can be inconsistent, even sometimes messy, but it needs to be accommodated.

It is not recommended practice to drive the bus at a speed outside the supported range of an attached peripheral. No argument. However, it is done because with some peripherals it does not cause a problem.

FWIW - even without a reified I²C bus in the API, a host has the information needed to implement the behavior you describe where all peripherals on the same bus communicate at the same, minimum speed.

fivdi, 2020-12-22:
>FWIW - even without a reified I²C bus in the API, a host has the information needed to implement the behavior you describe where all peripherals on the same bus communicate at the same, minimum speed.

Yes, I now see things this way too, in particular after seeing what can be done with [IO bus properties](https://github.com/EcmaTC53/spec/blob/master/docs/tc53.md#io-bus-properties) in #11.

closed 2020-12-22 by fivdi