EcmaTC53/spec #12: The `target` property

fivdi, 2020-12-22:
The Base Class Pattern defines its `target` property as follows:

The `target` property is opaque to the object's implementation. It may be initialized by the constructor using the `target` property in the options object. Scripts may both read and write the target property, though it is typically only set at construction.

Other than this, the spec doesn't have anything else to say about what a `target` is, so what is it? Given that a target is something that can be fired at maybe it's something for firing events at?

phoddie, 2020-12-22:
Alas, formal specifications are not great at making intent clear. 

The use of the `target` property is up to the script. It is unused by the implementation beyond setting the property on the instance during construction. If your code has a use for that, great. If not, it can be ignored.

As background, the intended use of `target` is to help with one of the recurring design challenges in ECMAScript -- managing `this`. The committee went through a few revisions on the topic before settling on the current approach.

The final design arrived at is simple and lightweight. The implementations of the classes in the specification  always invoke callbacks with `this` set to the instance. Client code can override that in the usual ways -- `Function.prototyp.bind` and arrow functions -- if they like. Clients that do not override it can use the value of `target` to get back to another instance that the callback functions needs to communicate with. The [SNTP example](https://github.com/Moddable-OpenSource/moddable/blob/399df5e022a187282fe6f7614b53be3ece532237/examples/io/udp/sntp/sntp.js#L17) is a trivial example of using `target` in this way.

fivdi, 2020-12-23:
Ok.

closed 2020-12-23 by fivdi