EcmaTC53/spec #23: Support accessing the remote address of a connected socket

cmidgley, 2022-04-22:
I have a use case where I need to run a server (eventually a WebSocket server when TC53 support is defined), where after a socket connection is established I would like to retrieve (if available) the address of the endpoint that made the connection.   The address information is to be used for logging purposes, to help debug and perform diagnostics on a complex environment with many connections.

I use this in a system today that shares a common JS source code base across Node and Moddable (with abstracted implementations for each platform).  In Node, when the web socket is connected, I am provided a handle to the request, from which I can receive the IP address:

```ts
wsServer.on("connection", (socket, req) => {
   req.socket.remoteAddress;  // this is the remote IP address
}
```

Using TC53, the concept of using the callback to notify on a new connection is different, where `onReadable` says one or more connections are available, and then `read` is used to get the socket (thereby making the callback itself an inappropriate place to put the remote address property).  Perhaps a solution here is to update the socket itself to provide a (read-only) property with the remote address.  In the case of a listen-established socket, it would contain the address of the client, and in the case of a client-established socket it would contain the address that the client connected to.  Just a suggestion - any way to receive the remote address of a listen-established socket would be appreciated.

Thanks for your consideration, and I look forward to seeing the server design proposals for the higher level protocols!

phoddie, 2022-04-22:
Thank you. This seems like a reasonable extension to `TCP`.  As you note, Node.js provides this today. The Moddable SDK provides the same functionality for its low level socket on some host platforms too.

phoddie, 2022-05-18:
Quick update. We discussed this issue today on the TC53 monthly call. There was consensus that it makes sense to support this in Ecma-419 2nd Edition. ]

phoddie, 2022-05-25:
Draft spec text is now in place. The Moddable SDK implementations of `TCP` for lwip and macOS [now both implement the feature](https://github.com/Moddable-OpenSource/moddable/commit/3892d052144d81a847c3ed5d11344062b342eee3).

closed 2023-01-07 by phoddie