EcmaTC53/spec #25: Conflict in the I2C spec/support multiple devices per bus

nebrius, 2022-06-09:
I'm going through https://419.ecma-international.org/#-10-io-classes-ic and working to implement it in https://github.com/nebrius/raspi-419, but I ran into something of a wall with the spec.

The I2C IO class constructor contains properties for data pin, clock pin, baudrate, address, and port, but this single constructor is confusing two separate concepts: I2C busses (data, clock, and baudrate) and I2C devices connected to these devices (address and port).

This is an issue because it's very common to connect multiple sensors to a single bus, which means that data, clock and baudrate _must_ be consistent across all three instances. In the case of the Raspberry Pi using [raspi-i2c](https://github.com/nebrius/raspi-i2c), this is especially difficult because the IO class has no concept of a sensor. The constructor takes in bus properties, and the write and read methods take in an address (port is not supported), meaning that the I2C instance from raspi-i2 maps to a bus not a sensor.

While it is theoretically possible to make this work using a lot of extra bookkeeping, it also means there are a lot more possibilities of user-error (e.g. when they supply a different `hz` value for one sensor compared to another on the same bus). This requires the end-user to write error handling code that would be unnecessary if we had a cleaner API surface.

This is especially annoying on the Raspberry Pi, which has only one I2C bus, meaning I'm having to implement error checking to make sure they specified the right pins, only to throw all of that information away and initialize the I2C bus with `new RaspiI2C({})

I recommend we take one of two approaches:
- More or less copy the API spec for raspi-i2c, which has a class with a constructor that represents a bus, with read and write methods that take an address and port as a parameter
- Create a class representing a bus, as above, and then pass that into the constructor for a second class that represents a device connected to the bus.

What do you think?

phoddie, 2022-06-09:
Thank you for the report. The committee spent time whether to expose the concept of a bus in the API. The v1 specification reflects the result of that. Let's see if we can find a comfortable solution within the framework of the specification.

The factoring of functionality across APIs in raspi-i2c and ECMA-419 are definitely different. That doesn't appear substantially messier to reconcile than what we've had to do various embedded implementations.

> ....Raspberry Pi, which has only one I2C bus, meaning I'm having to implement error checking to make sure they specified the right pins, only to throw all of that information away and initialize the I2C bus with `new RaspiI2C({})

Since your target device has a single I²C bus with fixed pins, the implementation could ignore `pins` and `port` on the options object passed to the constructor. Developers don't need to pass them for your device, since they are not configurable anyway.

Very often, developers using I²C don't specify `pins` or `port` and instead use the default I²C configuration (analogous to a default bus here) from the host provider instance. For example:

```
let s = new I2C({
    ...device.I2C.default,
    address: 0x20
    hz: 100_000
});
```

That works consistently across different devices without requiring the developer to be aware of the configuration. This was inspired, in part, by Boards in Firmata. It is the way most of the IO examples in the Moddable SDK are implemented so that they can run on any devices with I²C without requiring any configuration changes to the code.

> ...this is especially difficult because the IO class has no concept of a sensor

I'm not quite to understand the mixing of sensor with I²C that you describe. I²C is used for many purposes and sensors are just one of those. In ECMA-419 sensors are a layer above IO, one that can (and does) use various types IO (I²C, SPI, Digital, Analog, etc) depending on the requirements of the particular part.

>  ...it also means there are a lot more possibilities of user-error (e.g. when they supply a different hz value for one sensor compared to another on the same bus). 

That isn't necessarily an error. The question is whether the host implementation (a) supports the `hz` value they provide at all and (b) whether that `hz` can be used with the other active  I²C clients. (a) mostly depends on the hardware capabilities and (b) on whether the host implementation can switch clock speeds between active devices.

FWIW – I'm not very familiar with Raspberry Pi IO and that may be impeding this conversation somewhat. If I'm misunderstanding something, thank you in advance for your understanding in working through the details.

nebrius, 2022-06-09:
> Since your target device has a single I²C bus with fixed pins, the implementation could ignore pins and port on the options object passed to the constructor. Developers don't need to pass them for your device, since they are not configurable anyway.

What would you say to me throwing an error if a user specifies pins, and not including pin information in the default options? This would still require a change to the spec though, since the spec says these options are required. What I'm suggesting is:

```TypeScript
let s = new I2C({
    address: 0x20
});
```

I'm uncomfortable just ignoring what they pass in because this can confuse users if, say, they request I2C access on different pins than SDA0 and SCL0 but we operate on SDA0 and SCL0. IME the fact that I2C can only be used on certain pins has caused confusion with users who didn't realize at first that it's not like GPIO than can be used anywhere. Having explicit and descriptive error messages is really important in guiding users to working implementations, but just ignoring incorrect options is bad API design IMO.

If we set this up so that we threw an error saying "options.data and options.clock are not supported on the Raspberry Pi," then that directs users to remove those options.

> I'm not quite to understand the mixing of sensor with I²C that you describe. I²C is used for many purposes and sensors are just one of those. In ECMA-419 sensors are a layer above IO, one that can [...]

Sorry, I was using the word "sensor" colloquially, not in reference to what's in the spec (I haven't gotten that far yet!). What I should have said instead of sensor is "I2C device connected to an I2C bus"

> That isn't necessarily an error. The question is whether the host implementation (a) supports the hz value they provide at all and (b) whether that hz can be used with the other active I²C clients. [...]

Sorry but I must disagree with this statement quite strongly. IMO it's a very bad idea trying to run two devices at two different clock speeds on the same bus, even if it's technically possible to do so with clever coding in certain cases with certain devices.

The clock pin on a bus is wired directly to all devices, meaning that every device receives whatever clock signal is sent out over the line, regardless of the address specified. This clock is used to parse the address from the data line by every connected device (in a rather interesting open-drain wired OR configuration). In theory I2C devices should be able to operate on any clock signal sent in to them, but in practice there's a lot of shoddy I2C implementations out there that get finicky when a clock signal is out of range of what they want. The fact that these shoddy devices will see an alternating, interleaved clock signal can get them confused and effectively stop operating. Even well implemented devices will struggle if the clock speed set for the other device is higher than what this device supports.

Even if this weren't the case and all I2C devices perfectly support all clock speeds, this would be a major performance hit on the Raspberry Pi, because we'd have to destroy the I2C instance and close the I2C file descriptor after every write in order to reset the speed. Imagine sending a streaming TCP response to a remote client, except that we closed and reopened the connection after every packet.

phoddie, 2022-06-10:
> What would you say to me throwing an error if a user specifies pins, and not including pin information in the default options? 

That seems reasonable.

> Having explicit and descriptive error messages is really important in guiding users to working implementations

Agreed. And on a device like Raspberry Pi where you are free of the constraints of a microcontroller, it makes sense to be as helpful as possible.

> This would still require a change to the spec though, since the spec says these options are required

Understood. It is challenging in the spec to talk about what is required because hardware has so many variations. That detail will continue to evolve.

> IMO it's a very bad idea trying to run two devices at two different clock speeds on the same bus, even if it's technically possible to do so

I don't disagree. I also know that it is done. ECMA-419 needs to support reality to be meaningful to products.  As the introduction states: "Hardware can be inconsistent, even sometimes messy, but it needs to be accommodated."

From what you say about Raspberry Pi, it sounds like the clock speed is fixed at boot time. If I've understood that correctly, then `device.i2c.hz` will contain that value and the I2C constructor will validate it.

nebrius, 2022-06-10:
Thanks, that makes sense. I finished writing up the I2C constructor and default properties with what we talked about (and isn't spec-compliant as a result). Can you take a look and see if this matches up with your understanding of our conversation here? https://github.com/nebrius/raspi-419/blob/main/src/i2c.ts

phoddie, 2022-06-10:
I only noticed one issue in the constructor:

```
    if ((options as any).hz !== undefined) {
      throw new Error(`options.hz is not supported on the Raspberry Pi`);
```

Higher level objects (e.g. sensors) will pass `hz` as it is generally part of their configuration and because it is required. There is no reasonable default in the general case. This is in contrast to `clock` and `data` pins, for example, because a sensor generally takes those as inputs to its constructor so that its implementation is portable.

So, I think here you need to either ignore `hz` entirely (probably not ideal) or, when present, ensure that it matches the fixed `hz` of this boot of the device.

nebrius, 2022-06-10:
Ah ok, I'll make that change.

nebrius, 2022-06-10:
I think I'm done implementing everything in this issue, so I'm going to close it now. Next up is https://github.com/EcmaTC53/spec/issues/26, which is (hopefully) the only remaining blocker.

closed 2022-06-10 by nebrius