EcmaTC53/spec #32: Air Quality

Sineos, 2022-12-30:
The spec should also contain:
* HCHO (Formaldehyde): Although a subclass of the TVOCs, this is one of the most important poisons in living areas and in fact some sensors are able to distinguish between generic TVOCs and HCHO 
* PMx: Many of the particle sensors distinguish between PM0.5, PM1.0, PM2.5, PM4.0, PM10 whereas calibration is typically only done against the defined PM2.5 value

andycarle, 2022-12-30:
Thank you for the recommendations! We'll take a look at incorporating these into an upcoming revision to the spec.

Sineos, 2023-01-06:
One additional comment to https://github.com/EcmaTC53/spec/blob/master/docs/proposals/Sensor%20Classes%202022.md

Most of the commonly available (read affordable) gas sensors are not able to differentiate between different Alkanes, e.g. LPG, I-butane, Propane, Methane, alcohol etc.
Maybe a "combustible gas" class or similar would fit well

phoddie, 2023-01-07:
Interesting. If you could provide part numbers and/or links to some gas sensors that fit into your common/available/affordable group, that would be helpful.

Sineos, 2023-01-07:
Some example are the  MQ-x sensors for which many different variation exists with different sensitivities to various gases. 

### MQ-2
https://components101.com/sites/default/files/component_datasheet/MQ2%20Gas%20sensor.pdf

[image]

### MQ-135
https://www.electronicoscaldas.com/datasheet/MQ-135_Hanwei.pdf

[image]

Or some other sensor manufacturers:

### WSP2110 
https://www.winsen-sensor.com/d/files/wsp2110-(ver1_5)---manual(1).pdf

[image]

### MiCS-6814
https://www.sgxsensortech.com/content/uploads/2015/02/1143_Datasheet-MiCS-6814-rev-8.pdf

[image]

As you can see, some gases *could* be identified by very careful calibration to certain resistance ranges (something probably quite impossible without very specialized equipment). Other will not work at all, since their sensitive curves intersect or are very close nearby.